netmindz/balboa_GL_ML_spa_control #1: Message Format

netmindz, 2022-02-06:
Not yet sure which char is used as message delimiter or that bytesize and baudrate etc are defiantly correct.

Can't really start breakdown down the contents of the message until the basics are sorted

davewatson91, 2022-02-07:
OK, so I've had a very quick play around. 

If you split the file on 0xe1 20 20 20 80 0a 02 you can see 3x different message lengths (not sure if this is the right start sequence or not). 

I'm curious - would the spa have been 38.5c temperature when you took the dump? - possibly seeing that in ascii in each message (maybe what it's sending for the display - might change when you press buttons and have the display change while recording (which may be in the other dump files I haven't played with yet)).

davewatson91, 2022-02-07:
Further digging looks like that isn't a start sequence, the first byte in some packets appears to be possibly a tx/rx flag or user flag? - it changes when it looks like the temperature is changed. the ASCII does appear to be the temperature commanded, as it changes in the dumpfile dump-temp when split on 0xe9 as an end byte. 

I need better packet/byte analysis tools - does anyone have any suggestions? Wireshark isn't playing ball for importing for some reason, and isn't the greatest for the completely unknown protocols.

netmindz, 2022-02-07:
Yeah the spa was as 38.5 at the time with the temp dump was when i was trying to change the temp up to 40C

I should perhaps explain the dump files are not all the raw data, some of them were created from a script where i tried to guess the delimiter and write out each sequence on it's own line into the file to see if it looked plausible. 

CTRL+B looked like it might be right and would seem to make sense as logical value to choose as that means "start of text"

What is interesting is that I am not seeing a one2one relation to what is in the stream and the display. e.g when the display alternates between "Ecn" and the temp, while mode is Economy, there is no reference to Ecn as a a straight ascii string. Similarly when in the menus, no entries there.

Given on this board we have both 8pin Main Panel connectors and  6 pin Aux Panel connectors, I wonder how the differ in what they carry.

Lost access to the tub at the moment as the 9v pin I tried to used with stepdown to v5 to power the pi isn't working, was on usb charge pack instead that's now flat. Will have a look after work

netmindz, 2022-02-07:
dump-7 and dump-8 are raw

`python tcp_serial_redirect.py --bytesize 8 /dev/ttyUSB0 115200`
then
`nc hottub 7777 > dump-8`

Then the same, but bytesize = 7

netmindz, 2022-02-07:
See dumps/dump.cap for tcpdump of the data @davewatson91  - This includes me pressing lights on/off, pump1 and pump2 on and off and then dropping the temp down

netmindz, 2022-02-07:
@bggardner 

`FA@12332.511769
14@12332.522348
33@12332.522613
34@12332.522830
30@12332.523043
43@12332.523247
00@12332.523447
00@12332.523650
E1@12332.523822
00@12332.523990
00@12332.524155
00@12332.524317
80@12332.524479
02@12332.524641
FF@12332.524803
FF@12332.524966
5D@12332.525132
00@12332.525295
00@12332.525457
00@12332.525619
00@12332.525781
00@12332.525943
3D@12332.526108
AE@12332.531697
0D@12332.531867
00@12332.532032
00@12332.532196
00@12332.532359
00@12332.532521
00@12332.532682
00@12332.532844
00@12332.533006
00@12332.533169
00@12332.533330
00@12332.533494
00@12332.533656
00@12332.533818
00@12332.533979
2D@12332.534141
AE@12332.551847
0D@12332.552137
00@12332.552360
00@12332.552567
00@12332.552769
00@12332.552971
00@12332.553176
00@12332.553384
00@12332.553588
00@12332.553760
00@12332.553925
00@12332.554088
00@12332.554250
00@12332.554415
00@12332.554577
2D@12332.554740
FB@12332.554903
06@12332.555066
03@12332.555228
45@12332.555391
0E@12332.555555
00@12332.555718
00@12332.555881
FF@12332.556044
74@12332.556209
AE@12332.570718
0D@12332.570993
00@12332.571218
00@12332.571425
00@12332.571629
00@12332.571830
00@12332.572032
00@12332.572240
00@12332.572443
00@12332.572616
00@12332.572781
00@12332.572945
00@12332.573111
00@12332.573278
00@12332.573441
2D@12332.573604
FA@12332.592088
14@12332.592359
33@12332.592581
34@12332.592788
30@12332.592991
43@12332.593197
00@12332.593400
00@12332.593601
E1@12332.593804
00@12332.593978
00@12332.594144
00@12332.594310
80@12332.594473
02@12332.594635
FF@12332.594797
FF@12332.594959
5D@12332.595122
00@12332.595287
00@12332.595450
00@12332.595611
00@12332.595773
00@12332.595937
3D@12332.596099
FA@12332.610959
14@12332.611231
33@12332.611451
34@12332.611655
30@12332.611855
43@12332.612061
00@12332.612264
00@12332.612465
E1@12332.612665
00@12332.612836
00@12332.613001
00@12332.613167
80@12332.613330
02@12332.613492
FF@12332.613654
FF@12332.613816
5D@12332.613979
00@12332.614143
00@12332.614305
00@12332.614466
00@12332.614627
00@12332.614789
3D@12332.614952
FB@12332.615114
06@12332.615277
03@12332.615439
45@12332.615602
0E@12332.615764
00@12332.615927
00@12332.616089
FF@12332.616253
74@12332.616416
FA@12332.632337
14@12332.632609
33@12332.632831
34@12332.633043
30@12332.633248
43@12332.633452
00@12332.633655
00@12332.633858
E1@12332.634065
00@12332.634267
00@12332.634440
00@12332.634606
80@12332.634771
02@12332.634935
`

netmindz, 2022-02-07:
Tweaked to show time interval rather than time, see https://github.com/netmindz/balboa_GL_ML_spa_control/blob/main/tests/bggardner-timestamp-interval.log @bggardner

bggardner, 2022-02-07:
Looks like `FA` and `AE` are prime candidates, so maybe both are delimiters.  Dividing a dump up that way results in pseudo-triplicates like this:
```
AE0D010000000000000000000000005A
AE0D010000000000000000000000005AFB0603450E0000FF74
AE0D010000000000000000000000005A
FA14333835430000E1000000800A02FFFF65000000000050
FA14333835430000E1000000800A02FFFF65000000000050FB0603450E0000FF74
FA14333835430000E1000000800A02FFFF65000000000050
AE0D0A02000000000000000000000000C3
AE0D0A02000000000000000000000000C3FB0603450E0000FF74
AE0D0A02000000000000000000000000C3
FA14333835430000E1000000800A02FFFF65000000000050
FA14333835430000E1000000800A02FFFF65000000000050FB0603450E0000FF74
FA14333835430000E1000000800A02FFFF65000000000050
```
Notice that a byte sequence starting with `FB` is always after the 2nd "message" in the set.  I think we're getting somewhere.

netmindz, 2022-02-07:
yeah, that would appear to the the case,

See

- https://github.com/netmindz/balboa_GL_ML_spa_control/blob/main/tests/msg-count-light-on 
- https://github.com/netmindz/balboa_GL_ML_spa_control/blob/main/tests/msg-count-light-off

netmindz, 2022-02-07:
light of:

00000000: fa 14 33 33 35 43 00 **00** 6c 00 00 **00** 80 02 ff ff  ..335C..l.......
00000010: 5c 00 00 00 00 00 **d1** fb 06 03 45 0e 00 00 ff 74  \.........E....t
00000020: 0a  

light on:

00000000: fa 14 33 33 35 43 00 **03** 6c 00 00 **08** 80 02 ff ff  ..335C..l.......
00000010: 5c 00 00 00 00 00 **e4** fb 06 03 45 0e 00 00 ff 74  \.........E....t
00000020: 0a

netmindz, 2022-02-07:
pump1 on:

fa 14 33 33 35 43 **02** 00 6c 00 **04** 00 80 02 ff ff  ..335C..l.......
5c 00 00 00 00 00 **97** fb 06 03 45 0e 00 00 ff 74  \.........E....t
0a

pump2 on:

fa 14 33 33 35 43 **08** 00 6c 00 **08** 00 80 02 ff ff  ..335C..l.......
5c 00 00 00 00 00 **51** fb 06 03 45 0e 00 00 ff 74  \.....Q...E....t
0a

netmindz, 2022-02-07:
pump1 on:

fa 14 33 33 35 43 02 **00** 6c 00 04 **00** 80 02 ff ff  ..335C..l.......
5c 00 00 00 00 00 **97** fb 06 03 45 0e 00 00 ff 74  \.........E....t
0a 

pump1 on - light on:

fa 14 33 33 35 43 02 **03** 6c 00 04 **08** 80 02 ff ff  ..335C..l.......
5c 00 00 00 00 00 **a2** fb 06 03 45 0e 00 00 ff 74  \.........E....t
0a

netmindz, 2022-02-07:
pump1 on:

bggardner, 2022-02-07:
Let me rearrange so it's easier to decipher the byte sequence:

fa 14 33 33 35 43 00 **00** 6c 00 00 **00** 80 02 ff ff 5c 00 00 00 00 00 **d1** fb 06 03 45 0e 00 00 ff 74 0a // Light off
fa 14 33 33 35 43 00 **03** 6c 00 00 **08** 80 02 ff ff 5c 00 00 00 00 00 **e4** fb 06 03 45 0e 00 00 ff 74 0a // Light on

fa 14 33 33 35 43 **02** 00 6c 00 **04** 00 80 02 ff ff 5c 00 00 00 00 00 **97** fb 06 03 45 0e 00 00 ff 74 0a // Pump 1 on
fa 14 33 33 35 43 **08** 00 6c 00 **08** 00 80 02 ff ff 5c 00 00 00 00 00 **51** fb 06 03 45 0e 00 00 ff 74 0a // Pump 2 on

fa 14 33 33 35 43 02 00 6c 00 04 00 80 02 ff ff 5c 00 00 00 00 00 **97** fb 06 03 45 0e 00 00 ff 74 0a // Pump 1 on
fa 14 33 33 35 43 02 **03** 6c 00 04 **08** 80 02 ff ff 5c 00 00 00 00 00 **a2** fb 06 03 45 0e 00 00 ff 74 0a // Pump 1 on & light on

fa 14 33 33 35 43 08 **00** 6c 00 08 **00** 80 02 ff ff 5c 00 00 00 00 00 **51** fb 06 03 45 0e 00 00 ff 74 0a // Pump 2 on
fa 14 33 33 35 43 08 **03** 6c 00 08 **08** 80 02 ff ff 5c 00 00 00 00 00 **64** fb 06 03 45 0e 00 00 ff 74 0a // Pump 2 on & light on

It's easy to tell what the light and pump bits are.  The byte that always changes (byte 22) is most likely the checksum, so `fb` might actually be the start of another message (response?) since checksum is normally at the end.

netmindz, 2022-02-07:
We actually see two messages if you see msg-count-light-on, they start the same but one has an extra bit on the end. For the hex dump i went for the longer form, just in case in some cases it contained useful stuff and didn't want to mix or suddenly change message length

I've just committed the very hacky debug2.php that is able to read the status for light, pump1 and pump2, What is interesting us both pumps on not only changes the pump byte, but also the light swaps to 80

netmindz, 2022-02-07:
Runing `php debug2.php dump-panel-test` shows that the led byte is clearly dependent on other thing as seeing values of 04 and pump of c0

That was a dump taken of where i basically worked my way through all the buttons to see if i could grab all the different messages. running the script like that allows you to replay the events, though limited use i guess without a video of what i was doing at the time

netmindz, 2022-02-07:
For reference the following

fa 14 33 33 35 43 

is the header, (fa 14) then 335C in ascii when the tub us at 33.5C

netmindz, 2022-02-09:
Have just pushed an updated  debug2.php script 

Making good process with showing status of the tub - got most stuff mapped now, bar knowing ecn/slp/std

netmindz, 2022-02-09:
I've also tried to summarise on the wiki https://github.com/netmindz/balboa_GL_ML_spa_control/wiki/Message-Format

I'm getting the feeling that there is some bitmask style stuff is going on or something similar as things like while i thought was just the light, seems to change when you turn on multiple pumps

netmindz, 2022-02-09:
Thanks for the great tip on using timing as clue to delimiter @bggardner and the help with the `FA14` messages.

Any ideas on the `AE0D` messages or why you think we see the repeating format of messages i describe on the wiki page?

netmindz, 2022-02-09:
@davewatson91 - Did you see I have a issue to track the pinout and the use of the remaining pins? https://github.com/netmindz/balboa_GL_ML_spa_control/issues/2

If you could take a look at the scope plots to see if you have any ideas about the rather odd values that would be much appreciated

bggardner, 2022-02-09:
Try recording messages (with timestamps) during initial power-on of the spa.  At this point, maybe only record timestamps per-message instead of per-byte.  Do this with the top panel/display connected and disconnected and see if there is any different traffic.  You can also try disconnecting the top panel/display while it's running, but be very careful as there is high voltage present.

netmindz, 2022-02-20:
@bggardner without the top panel connected there is no traffic at all on the rs485 link. I wonder if the other pins form some part of discovery/negation system.

Have folk looked at the scope plots for the other pins. Any guesses at to what they are?

netmindz, 2022-03-04:
As only 3 pins, that to me makes me think I2C maybe, but then what about the extra pin?

netmindz, 2022-03-05:
I've committed the first very very rough PoC that is able to read the temperature data and show up automatically in Home Assistant

netmindz, 2022-03-05:
So, bit more info. If you unplug the top panel then all data stops on the rs485 line, so either this is creating the data to be shared with other devices - say long distance link for aux panels inside the property based on the other pins or that the data we are seeing is only triggered by the panel with the control unit responding.

netmindz, 2022-03-12:
[image]

Have an ESP8266 now running feeding the status data to Home Assistant. So thanks for everyone's help getting this far, next challenge is going to work out how to actually send commands, so any help would be much appreciated on that, see the other thread for discussion on the data lines

netmindz, 2022-03-12:
@bggardner - I think you may be right about the FB 06 being some form of reply to the FA 14, what is odd about that though is why would you only get that after the first FA 14.

Turning my attention to the AE 0D messages, they actually cycle between AE 0D 00, AE 0D 01, AE 0D 02 and AE 0D 03

I'm also seeing some messages that are CA 82

Does anyone have a spare ESP8266 or ESP32 and RS484 adapter to try connecting up to their tub too? Have got to the stage that is actually useful for practical purposes, like checking up to temp from the comfort of the sofa and seeing the graphs and logs in Home Assistant for how the wind triggers the heater more often etc

tmjo, 2022-03-12:
> Does anyone have a spare ESP8266 or ESP32 and RS484 adapter to try connecting up to their tub too? Have got to the stage that is actually useful for practical purposes, like checking up to temp from [...]

@netmindz; I'm impressed by your progress here, so very much so that even if I have trouble finding the time to prioritize this I will try to find a moment and give it a try! :) It is really inspiring! If I get it right, I should try your code on an ESP with RS485 connected to pin 1+3 and then look for mqtt messages in HA right?

netmindz, 2022-03-12:
@tmjo Thanks, it's been quite the learning curve, blind hackery and looking at all the great work done already on the other project for inspiration. 

Yeah, you need to use Pin 1 and Pin 3 to an RS485 adapter

Sorry I haven't made a note of which pin is A vs B - the wrong way round gives data, but garbled.

If you use my sensor sketch, at the moment it's setup either for either ESP32 that connects via TCP/IP to a RPi using https://www.amazon.co.uk/gp/product/B016IG6X7I/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1 and a script to bridge serial to network so I can sit in the warm, with the ESP32 connected to laptop so i can see the Serial debug

OR

If you compile for ESP8266 then it expects to have RS458 connection on pins D6 and D7. I'm using https://www.amazon.co.uk/gp/product/B07DN115BZ/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1 Either needs to be this one the older style with the DE/RE pins set to be in receive mode. Still best to check Serial debug to see all is ok.  In this setup though it's harder to just swap to the raw data stream. I've started adding a telnet server to it so i can also get a copy of the raw hex in realtime, but that's a bit flaky, so probably advise against trying to use that right now

The file contains a comment showing how to create a wifi.h file in the sensor directory with your WIFi details. You will also need to update the file of course to put the IP of you MQTT server and add login details if your requires them

The device should then appear as new device in HA with auto config of the different sensors

I'd suggest starting out with just seeing if you can get the sensor app onto an ESP and showing up in HA, shouldn't need any special hardware for that. If you are unsure what libraries are needed, see my Arduino repo. Once that is working, connecting the hardware up can be a project for another day. Pin 6 does have enough power for a 5v adapter for ESP,  but not for RPi

netmindz, 2022-03-13:
My ESP8266 died so the TCP or direct serial connection now it's own define.

ESP32 set to use hardware serial for better reliability

Simple status webpage added for quick view of the tub without Home Assistant

tmjo, 2022-03-14:
I tried poking around a bit in the weekend, but can't get it to download to my ESP. Code is compiling fine and I found the libraries you used. Believe the problem is either a dead ESP, or a problem with versions here or there after I changed PC some time ago. It doesn't seem to find the COM-port, although I know it is the right one and the ESP even starts to blink when trying to download. I have a feeling I've seen that before and solved it by downloading an older ESP-version, but that didn't work this time. I'll see what I can do, or find another ESP to check if it is working.

netmindz, 2022-03-14:
Glad you managed to compile ok. Do you have the right windows driver for the serial interface for that esp? It can vary between boards as what is paired with the ESP

netmindz, 2022-03-14:
P.s don't forget to pull the latest commits into your fork

tmjo, 2022-03-14:
Right, it could be drivers or even the cable itself. But it worked before though. I'll some other options, or even a different computer to see if I can get it to work.

tmjo, 2022-03-14:
Oh, and would you mind linking these two libraries? I pulled last commits, but wasn't specifically which libraries these are:

```
#include <WebSocketsServer.h>
#include <WebServer.h>
```

netmindz, 2022-03-14:
> Seems it is my ESP board, I found another one and it works - download complete. But now I need to do some minor soldering to move my RS485 board to this ESP. The RS485 I have is [this one](https://www.ebay.com/itm/381374599127) I think and I had it wired to TX/RX on the ESP while you use D6/D7. Do you think my board will work with D6/D7 too, or should I stay on TX/RX?

For that controller, you need to set pull RE DE high/low for when you what to send vs receive. At the monent we are only reading, so RE is the 'Receiver Enable' pin and must be pulled low whenever you want to be able to receive data - so connect to gnd.

https://github.com/Links2004/arduinoWebSockets

The WebServer.h was just from the ESP platform examples

netmindz, 2022-03-15:
To save spamming folk too much with notifications from my chain of thought, I'm going to put my thinking here https://github.com/netmindz/balboa_GL_ML_spa_control/discussions/3 and keep this issue more for status updates, key breakthroughs etc

netmindz, 2022-03-16:
I have fixed the issue with sometimes getting incorrect status data, e.g looking like we pump is running when it's not for a brief period. 
[image]
[image]
[image]

netmindz, 2022-03-20:
Just tagged first release and published binary, but just released that it will have my wifi details set and no way to change, so not much use, but a key milestone none the less that it's at a point that it's actually useful now

netmindz, 2022-03-20:
Updated release with fallback. If you want to just flash this binary to ESP32 without messing about with compiling, then you can see the current temp of the tub on http://192.168.4.1/ 

https://github.com/netmindz/balboa_GL_ML_spa_control/releases/tag/v0.0.6-ap-fallback

netmindz, 2022-03-22:
How did you get on @tmjo ?

tmjo, 2022-03-22:
Hi! Sorry, I didn't get around to complete my tests during the weekend, but I'll try as soon as I can. We're having a baby one of these days, so for some reason wifey expects progress on some other stuff as well these days. :D

netmindz, 2022-03-22:
Oh, congratulations. Exciting times!

tmjo, 2022-03-25:
Thanks! I got everything compiled now and it's up and running in Home Assistant and the webserver, but still pending to solder the RS485 and eventually go out and connect it to my tub. But looking promising so far.

I've never used ESP32, only ESP8266, so I found out I had to modify WebServer.h to ESP8266WebServer.h and WiFiAP to ESP8266WiFiAP.h for it to find the files and compile. I also added credentials to the mqtt connection, but then it all appeared nicely in HA.

Who knows, maybe this weekend I'll test my luck and see if I can get the real thing going :) Thanks for your work @netmindz !

netmindz, 2022-03-26:
That's great news @tmjo .  Thanks for the feedback about the esp8266, I'll move those includes to the right ifdef block

netmindz, 2022-03-26:
Latest version adds status info, current mode and target temp as well as recording events like temp+ and temp-

First hints as to how we can actually *send* a message, once know what the other values are

tmjo, 2022-03-26:
Sounds good! 

I gave it a shot outside today, but unfortunately I didn't have any luck and couldn't receive any status from my tub. I'll review my circuit, and see if I can get hold of another RS485 board just in case it has been damaged. I also forgot to add the RE/DE pin to the program, but it would perhaps be safe to assume that it is default low and therefore RX?

Another thing that came to my mind just to be sure: Pins 1+3 would be as indicated in the image? That's how I numbered them when I did my initial measurements, but I don't know if it is the correct way, so would be good to confirm if you're using the same numbering :)

[image]

Do you happen to know an easy way to verify if my ESP+RS485 is working correctly? I guess if I had two, I could let them talk to each other, but right now I only have this one setup.

Anyway, good to see you're making progress, I'm sure I'll get there too eventually!

netmindz, 2022-03-27:
Sorry to hear you didn't have any luck. Yeah i'm using your numbering scheme 

If you are not using the newer RS485 adapter I list, then you *must* follow the instructions regarding the DE+RE pins or they will not work properly. Lost so much time when working on DMX stuff due to incorrect usage of the MAX485 chips

netmindz, 2022-03-27:
As we are only reading data, DE/RE both need to be pulled low

tmjo, 2022-03-27:
Ok, so I have some progress! I've got RE and DE jumpered together on the board since I've seen that's quite common, and added code to set the output low. Now I got some data flowing, although there was not any sense to the data. I assumed it was the RS485 polarity and tried to swap A/B but still not anything that makes sense. 

Here's a screenshot from HA
[image]

I know you said it wasn't all stable yet, but I tried to Telnet into the device, and there is certainly something coming:
`Hello 0
       ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d03000000000000000000000000b4fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d03000000000000000000000000b4fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d03000000000000000000000000b4fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d03000000000000000000000000b4fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d03000000000000000000000000b4fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d03000000000000000000000000b4fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d03000000000000000000000000b4fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d03000000000000000000000000b4fb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d03000000000000000000000000b4fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d03000000000000000000000000b4fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d03000000000000000000000000b4fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
[... 110 lines cut ...]
ae0d03000000000000000000000000b4fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d03000000000000000000000000b4fb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d03000000000000000000000000b4fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d03000000000000000000000000b4fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
ae0d03000000000000000000000000b4fb060343060000ffce
ae0d000000000000000000000000002dfb060343060000ffce
ae0d010000000000000000000000005afb060343060000ffce
ae0d02000000000000000000000000c3fb060343060000ffce
`

netmindz, 2022-03-27:
So interestingly you aren't seeing the same pattern of 3 groups of FA messages and then the 3 groups of AE messages that I see. Do you know the model of your top panel?

We are seeing the AE0D then 0,1,2,3 messages, which for you as me are basically all 0 then the FB ending

Was this running the latest 0.0.8 version? 

The app includes OTA support, so you should be able to upgrade from the comfort of desk/sofa rather than usb to the controller outside. If you do connect up the usb, can you grab some of the serial debug output too?

Can you also share the output from some button presses, e.g light on and off please @tmjo ?

netmindz, 2022-03-27:
For the messages see https://github.com/netmindz/balboa_GL_ML_spa_control/discussions/3

tmjo, 2022-03-28:
My top panel is a ML400
[image]

I'm excited that I get some data, it's at least a good step in the right direction. I was still running the 0.0.7 version, but I'll try to update. OTA is nice, would make it easier. I'll see if I can get some serial debug from console and I think maybe we've got some RS485 adapters at work to get it directly to PC too.

I'll also try to cycle power on the ESP after swapping RS485 polarity. Not sure if it matters, but some guys at work said they'd seen cases where it was necessary like if the electronics were hanging after wrong polarity in rare cases. If I remember correctly, I didn't see any change in data when I swapped, so it could be the case. Worth a shot anyway.

I see you mentioned tub pin 6 is capable of powering an ESP8266 board, are you running that setup successfully? Would be great to leave the ESP there in combination with OTA update.

netmindz, 2022-03-28:
The polarity looks correct to me, While some of the messages a different, e.g 2131 in the FA messages where i see FFFF, otherwise looks generally right. 

It was unlclear from your example how long you had the telnet session running for, what state the tub was in or if you tried making any state changes in that time period.

As for pin 6, yeah that was not able to power a Raspberry Pi, but can power the ESPs. Just be sure to use the proper voltage adapter to step down to the properly regulated 5V!

tmjo, 2022-03-28:
Hmm, yeah, the Telnet log was a bit random to be honest, but the tub was in heating mode and water temperature probably somewhere around 7-12C as I used the opportunity to change the water and give the tub a spring cleaning. I was inside on my computer, so no interaction with the top panel at the time of logging.

The mosquitto_sub command you mentioned above, where do you run that? In a Home Assistant terminal? I'm not familiar with the method, but if I understand google right it's a call included with mosquitto which makes a log file?

I saw your warning about using the 5V directly on the ESP, but mine is a Wemos D1 Mini and has a 5V supply pin. I've seen different opinions on using 5V directly on inputs though, so that is probably good to avoid, but shouldn't it be ok to power the 5V supply directly? Or is it too unstable? Anyway, I don't need to mess with that now before I get it running.

tmjo, 2022-03-28:
Ok, so got inspired and did some testing out in the cold again. I can't say I'm confident about what I'm doing, so please have patience with me :) But I believe it's getting somewhere.

Take a look at some [testfiles](https://github.com/tmjo/balboa_GL_ML_spa_control/tree/test/tests/tmjo/2022-03-28). The capture-files are from Realterm-software listening to a RS485 adapter directly to my PC while the msg-files are mosquitto_sub files from Home Assistant and the png files are the grep (could copy the text, hence the image file).

I don't seem to get any reasonable data to HA, and it doesn't seem to update very much. Not sure if it is my hardware or what it is. But looking at the capture files I feel that I'm seeing something interesting. `capture.txt` seems to be garbage and does not contain any FA14 headers. `capture2.txt` and `capture3.txt` on the other hand do and is where I switched polarity. Stupid as I am I can't remember when I swapped polarity, but seems like it was in the middle of `capture2.txt`. I'm sure it was changed in `capture3.txt`. Sorry for the confusion, but it was cold outside :)

At the time of capturing, the temperature was around 30C. I switched on the light and back off again, started pump and back off again. I didn't study the result before coming into the warm again, but looking at `capture3.txt` I see

```
FA14 333035 43 0 0 1 06100000080021212560000000000A5 FB060343060000FFCE
FA14 333035 43 0 0 1 06100000080021212560000000000A5
FA14 333035 43 0 0 1 06100000080021212560000000000A5
```

which should correspond to 30,5C right? Then I see some AE-messages:

```
AE0D1648494071,
000000000000000000000000002DFB060343060000FFCE
AE0D1648494071,
000000000000000000000000002D
AE0D1648494071,
000000000000000000000000002D
```

I am not sure about the comma in my logfiles. It could be the Realterm software adding some info as I tried to activate timestamping.

But this is looking promising, right?

netmindz, 2022-03-28:
Yeah with the right polarity you are now getting 30.5C

netmindz, 2022-03-28:
There are two different voltage warnings, the pins in not using (yet) are 5v and I suspect I2C, then these need level shifter

The pin I'm using for power in the one that is the rather random 9.7volts, so needs to be stepped down

I'd say if you swap back to the esp with alternative polarity you should be up in action as you are seeing the FA14 messages now, which is basically where all my attention has been so far

tmjo, 2022-03-29:
I'll dig more into it. The ESP was connected to the same circuit as the RS485-dongle with the same polarity, that's what I found a bit strange. Seems I get correct info on the dongle but not on the ESP, so it may be that my hardware is not good. Forgot to check the arduino serial monitor, I'll do that next time.

[68 more replies]

closed 2023-05-05 by netmindz