netmindz/balboa_GL_ML_spa_control #2: Pin usage

netmindz, 2022-02-06:
Pin 2, 4 + 5 have unknown format / purpose 

Anyone have any guess?

netmindz, 2022-02-09:
What from the previous diagnostic was listed as 9.7V which I tried to connect a stepdown converter to get 5V to feed my Rpi, the Pi was failing to boot, so checked the pin i'm using and now seeing 5V coming from the tub. Not sure if just never attached the scope or multimeter during my tests as we thought we knew that pin already or it's somehow randomly changed voltage. I did check vs both 0/GND pins just in case i saw a difference with the scope

netmindz, 2022-02-20:
Tested again with multimeter rather than scope and that was showing the 9v again, so not sure if I was doing something wrong with the scope or if the voltage charges under circumstances

netmindz, 2022-02-20:
This post might be of use. Now I've found my other scope probe I'll try and get some captures of pairs of pins to see if I get more like he does and try pressing buttons too 

https://www.olivierhill.ca/archives/72-The-Internet-of-Spas.html

netmindz, 2022-03-20:
Warning - don't be a Muppet like me and connect any pins directly to the ESP as you will blow it up as it's not 5v tolerant

netmindz, 2022-09-16:
Just had an idea @tmjo  - what if the spare pins aren't a different protocol, but actually just indicate when a device on that port is allowed to transmit data to the rs-485 bus?

Might be an idea to log change of state Vs when it's a message from top panel or control unit we are seeing

netmindz, 2022-09-16:
Would be more reliable than just replying on timing to intersperse the messages.
Might also be worth checking if we see any difference on the port where we have real topside panel Vs or esp connected

tmjo, 2022-09-16:
Interesting thought, it could make sense!

netmindz, 2022-09-17:
The "spare" pins are indeed I2C

I2C device found at address 0x04
I2C device found at address 0x13
I2C device found at address 0x22
I2C device found at address 0x31
I2C device found at address 0x68

netmindz, 2022-09-17:
or maybe not - now not seen any devices when I run the WireScan example

netmindz, 2022-09-17:
Or maybe not - getting different number of devices found each time the scan runs

netmindz, 2022-09-17:
So, I hooked up the required level converter to go from the 5v of the tub to the 3.3V of the ESP

I initially tried jumping straight in and recording the value of pin 5 vs certain messages, to see for example if pin5 was low during the transmission by the top panel (the "long" messages)

This was not very consistent. Sometimes looking like there was a relationship, then seeing different results

I played round with a few different places to read that value, e.g adding an else to the bytes variable check. No luck

So I then did a much more basic check, to keep reading the RS485 data until we saw the pin go LOW, then print want we have. My thinking being that if pin 5 is in any way related to the RS485 messages then we would see some form of logical grouping, e.g always being just before the second half of a long message. Sadly all I got looked very random. Sometimes no message at all, sometimes very long strings with many many messages, sometimes much shorter groupings. No real pattern that I could see.

I then tried disconnecting the top panel while the system was running and then i saw much more consistent value for pin 5 vs where we are in the message stream

e.g

> 11:19:50.677 -> message = ae0d000000000000000000000000002d digitalState = 0
> 11:19:50.677 -> message = ae0d000000000000000000000000002d digitalState = 1
> 11:19:50.677 -> message = ae0d000000000000000000000000002d digitalState = 1
> 11:19:50.678 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 0
> 11:19:50.678 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 1
> 11:19:50.709 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 1
> 11:19:50.709 -> message = ae0d010000000000000000000000005a digitalState = 0
> 11:19:50.742 -> message = ae0d010000000000000000000000005a digitalState = 1
> 11:19:50.742 -> message = ae0d010000000000000000000000005a digitalState = 1
> 11:19:50.775 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 0
> 11:19:50.809 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 1
> 11:19:50.809 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 1
> 11:19:50.842 -> message = ae0d02000000000000000000000000c3 digitalState = 0
> 11:19:50.842 -> message = ae0d02000000000000000000000000c3 digitalState = 1
> 11:19:50.875 -> message = ae0d02000000000000000000000000c3 digitalState = 1
> 11:19:50.908 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 0
> 11:19:50.908 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 1
> 11:19:50.941 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 1
> 11:19:50.941 -> message = ae0d03000000000000000000000000b4 digitalState = 0
> 11:19:50.974 -> message = ae0d03000000000000000000000000b4 digitalState = 1
> 11:19:51.008 -> message = ae0d03000000000000000000000000b4 digitalState = 1
> 11:19:51.008 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 0
> 11:19:51.041 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 1
> 11:19:51.041 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 1

So at first glance, you might say *bingo*, this lines up with the port-based scanning and we transmit when it's low, but my ESP is connected to the 2nd port, so I would expect 010 not 011.  You could say it's 001 because during initialisation I had my real panel in the first port, so maybe it doesn't bother going low if it thinks nothing on that port. 

however, see here how we seem to skip the the FA messages and all of a sudden it's now 110 - low for final message in the triplet

> 11:20:46.619 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 0
> 11:20:46.652 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 1
> 11:20:46.652 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 1
> 11:20:46.685 -> message = ae0d03000000000000000000000000b4 digitalState = 0
> 11:20:46.718 -> message = ae0d03000000000000000000000000b4 digitalState = 1
> 11:20:46.718 -> message = ae0d03000000000000000000000000b4 digitalState = 1
> 11:20:46.752 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 0
> 11:20:46.752 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 1
> 11:20:46.785 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 1
> 11:20:46.818 -> message = ae0d000000000000000000000000002d digitalState = 0
> 11:20:46.818 -> message = ae0d000000000000000000000000002d digitalState = 1
> 11:20:51.734 -> message = ae0d000000000000000000000000002d digitalState = 1
> 11:20:51.768 -> message = ae0d000000000000000000000000002d digitalState = 1
> 11:20:51.768 -> message = ae0d000000000000000000000000002d digitalState = 0
> 11:20:51.801 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 1
> 11:20:51.801 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 1
> 11:20:51.834 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 0
> 11:20:51.867 -> message = ae0d010000000000000000000000005a digitalState = 1
> 11:20:51.867 -> message = ae0d010000000000000000000000005a digitalState = 1
> 11:20:51.900 -> message = ae0d010000000000000000000000005a digitalState = 0
> 11:20:51.900 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 1
> 11:20:51.933 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 1
> 11:20:51.967 -> message = fa14333830430000610000008002ffff640000000000d4 digitalState = 0

With the top panel I think the state was even more erratic.

Wondering if it's worth testing with striped down test in case we are missing the LOW pulse due to delays by the other bits in the main loop?

netmindz, 2022-09-17:
Bing!

https://github.com/netmindz/balboa_GL_ML_spa_control/blob/pin5/pin5test/pin5test.ino

No hacky reading of data based on bytes and length, read based on the clock

tmp = fa14333735430000690000008002ffff640000000000f4fb0603450e0000ff74
tmp = fa14333735430000690000008002ffff640000000000f4
tmp = fa14333735430000690000008002ffff640000000000f4
tmp = ae0d02000000000000000000000000c3fb0603450e0000ff74
tmp = ae0d02000000000000000000000000c3
tmp = ae0d02000000000000000000000000c3

netmindz, 2022-09-17:
Doesn't resolve my issue of when i can send my own FB sequence however as if I want for pin 5 to go LOW before i handle the message, then i'm too late to send my FB

netmindz, 2022-10-29:
I'm wondering if to solve the timing issue, that I would be better using an interrupt so that the send of the message is triggered as soon as the pin changes state

netmindz, 2023-03-16:
The reply must be sent before the pin changes state, so we are going to need to read a fixed number of bytes based on what we read when the pin first changes, e.g FA

TiSiAi, 2023-03-16:
Yes, the response must be send juste after the end of the FA14 message and while the pin 5 is low.
the pin is low about 3ms
the FA14 frame is less than 1,5ms
On this picture, you can see a little pause and after de FB06 command
On my code I put a timer on receive data and when I have about 200us without data after a FA14 command, I can send
But no test begin, I finalise my sample code

[image]

netmindz, 2023-03-16:
Yeah I'm happy we are confident about pin5 purpose, just 2 and 4 that are unknown, but possibly something we don't need?

TiSiAi, 2023-03-16:
;)
The pin 2 is to power the backlight on ma panel ( not the LCD and button, another backlight )
the pin 4 is not a DC ( but alternate ) 5V but I don't know for the moment the utility

netmindz, 2023-03-16:
I'll have to retest pin2 without my top panel and see if what I see on the scope changes then from the really odd pattern I have currently

netmindz, 2023-03-18:
Yeah, you are seeing the same odd dip as me on input1 vs input2.

Could you try with pin5 and the output from the rs485 adapter as well? just to totally rule out these indicating RTS/CTS style information

TiSiAi, 2023-03-18:
Pin 5 in yellow
Output RS485 adapter in blue

We have less than 3ms to receive the data and response if we need to send a command

OUTPUT RS485 ( rx data and tx data in my sample of the keyboard because I use the PIN 5 of the keyboard to analyse frame ).
I don't link the TX data of my RS485 converter.

[image]

below , 3 frame ( 1 for each keyboard ) > ignore them when the pin5 is not low.

[image]

Below PIN5 of channel keyboard 2 et PIN5 of channel 3
[image]

netmindz, 2023-03-21:
@TiSiAi - Please see https://github.com/netmindz/balboa_GL_ML_spa_control/pull/13 for my attempt to read data when we see pin5 go low and then try to be ready to send our data before the pin goes HIGH again

netmindz, 2023-06-16:
Going to close this issue as not sure we will ever get to the bottom of pin2 and pin4 and at the moment, they aren't actually blocking us going forward right now

My best guess at the moment is CTS / RTS, but don't have scope data from @TiSiAi  showing these 2 pins vs the rs485 

https://en.wikipedia.org/wiki/RS-232#RTS,_CTS,_and_RTR

closed 2023-06-16 by netmindz