netmindz/balboa_GL_ML_spa_control #12: GS510SZ - Control system

MagnusPer, 2022-08-15:
Long shoot but want to ask about the support of [GS510SZ](https://www.tubs.parts/en/balboa-gs510sz-control-system-p-434.html) control system. I have been searching a lot for anyone started to decode the data for this control system but have not stumbled over any work yet. My question for this repository is if this development for the GL/ML Balboa system is compatible?

I have programming experience so I can absolutely contribute here!

tmjo, 2022-08-16:
Hi! I'm sure @netmindz will fill in with more details too, but from my point of view I would be very surprised if there are more than two types of communication within (relatively) so few years. So either this repo should do the trick, or [this repo](https://github.com/ccutrer/balboa_worldwide_app) for the newer models. There's also an [ESP-version](https://github.com/cribskip/esp8266_spa) for the newer models.

You plugs, are they RJ45-type or Molex-type? Perhaps if you are able to figure out if your spa is compatible with the [Balboa Wifi-module](https://www.boospa.net/en/balboa-wi-fi-module.html), I believe that would give a hint. If it is, I would give it a try with the other repos but if it is not compatible, I would assume it is same or similar to this repo we are working on here.

Would be great for us if your spa is compatible with netmindz code here, because it would help to further reverse-engineer the protocol :)

MagnusPer, 2022-08-16:
Thanks for answer.
I'm sure the Balboa GS control system not supporting the [Balboa Wifi-module](https://www.boospa.net/en/balboa-wi-fi-module.html). Also it was mentioned in the repo[ balboa_worldwide_app](https://github.com/ccutrer/balboa_worldwide_app) that GS variant not was supported.

My plug to the control panel is RJ45, see [VL700S](https://www.tubs.parts/en/balboa-vl700s-control-panel-p-836.html)

tmjo, 2022-08-16:
Sounds good, then I would bet that your spa fits to netmindz code here. I have Molex-plug, and I think he has too. So first step in my point of view would be to measure the pins of your RJ45 and try to identify which pins are the communication pins. For the Molex, you find some information in [the Wiki](https://github.com/netmindz/balboa_GL_ML_spa_control/wiki/Install-Guide). You also find some measurements I did in [this thread](https://github.com/ccutrer/balboa_worldwide_app/issues/14). If I remember correctly, this repo was using a RJ45-plug so perhaps there is some useful info there too.

Once you've got an idea on which pins are the communication (or at least not the supply pins to avoid burning the ESP), I would suggest just downloading the code and put it onto a ESP8266 or ESP32 (preferably the 32,  but I'm running mine on the 8266 just fine) with a RS485-adapter to go with it. 

Are you running Home Assistant or do you want to communicate to you spa in different ways? This repo for now talks to Home Assistant through MQTT, but it would be a small job to convert it to plain MQTT.

tmjo, 2022-08-16:
I remember seeing that site back when I was googling around too. It could be useful. We haven't yet figured out what all the pins are (for Molex), but it could be similar (although we also read light, pump, clock and temp on the serial communication).

MagnusPer, 2022-08-16:
This [site ](https://www.olivierhill.ca/archives/74-Cal-Spa-Connector.html)may give a hint on the RJ45 pinout

I do not use Home Assistant I'm a [Homey ](https://homey.app/sv-se/) and [NodeRed](https://nodered.org/) user and prefer plain MQTT. But as you say its the small work to convert it later

netmindz, 2022-08-17:
As you have more buttons, I don't think that other guide will be correct for you.

As @tmjo has already mentioned, first thing would be to establish what's on what pin. Basic voltage test should give some indication, if you don't have access to an oscilloscope

MagnusPer, 2022-08-17:
First thing for me is to order some components and then after try to connect to see if I get some data. I can borrow an oscilloscope to first check the different pin usage.

Question - Did you connected by Y-cable between controller and panel?

netmindz, 2022-08-18:
The control unit we both have supports up to 3 main panels and 2 aux panels, so we just use one of the main panel spare ports

netmindz, 2022-08-20:
Any luck identifying any of the pins @MagnusPer ?

MagnusPer, 2022-08-21:
Have ordered parts and prepared Y-cable, next step is to borrow a oscilloscope and then after I will start to do my first measurements. Coming back here soon with feedback...

MagnusPer, 2022-08-24:
Have done the first oscilloscope measurement to find correct pins in the RJ45 connector and its easy to see changes when pressing buttons when connecting between PIN4 and PIN 5. Seems that this [site](https://www.olivierhill.ca/archives/74-Cal-Spa-Connector.html) describes the correct pin configuration for the GS510SZ (_However this needs to be analyzed further_)  

PIN 4 = GND
PIN 5 = DATA 

[image]

**Question:** Are you then connecting the RS485 adapter to those pins?

MagnusPer, 2022-08-26:
Now I have measured the behavior on the different RJ45 pins and my conclusion is that the GS510SZ has a totally different construction/design. Its more like described in [this](https://www.olivierhill.ca/archives/74-Cal-Spa-Connector.html) site. 

It has no RS-485 interface it's a serial interface which I need to decode.  

PIN 1 = 5V
PIN 2 = Digital Input for ??
PIN 3 = Digital Input for Pump 1
PIN 4 = GND
PIN 5 = Tx-DATA from display - 0-5V
PIN 6 = Clock
PIN 7 = Digital Input for Pump 1
PIN 8 = Tx-DATA for Temperature button (when pressing up/down temp then )
 
**Question** - Anyone knows if [Oliver ](https://www.olivierhill.ca/archives/74-Cal-Spa-Connector.html)shared the code anywhere?

netmindz, 2022-08-28:
Ah, that's bad news.  I had hoped while the physical connection was different, it might share the same serial protocol.

Could you provide more information as to how you had come to your conclusion regarding which pin is for what purpose.

If they are using a system more like Oliver discovered, the process of trying to explain what you have discovered so far might bring further insight, so happy to bounce ideas

MagnusPer, 2022-08-28:
@netmindz - started to document my measurements and initial conclusions here - https://github.com/MagnusPer/Balboa-GS510SZ

I trying to decode the bitstream to my display value of 37.5C but have not yet figured out how. If you have some time over I would appreciate that. Have tested both LSB and MSB also inverted and not but...

netmindz, 2022-08-30:
Yeah I think you are correct. For the GS series the connection does seem to be the top panel is just a function of the main control, so the data being sent is just the mechanism for driving the display etc

Whereas the GL series, with it's ability to have 3 main panels and 2 auxiliary panels, means more of a communication network between all of these different devices

closed 2022-09-10 by netmindz