netmindz/balboa_GL_ML_spa_control #20: ESP32 code problem

mbuhansen, 2023-04-27:
I have got a ESP32, and try to use your code, but in validate i get error when i use it. Can you help me i am new with ESPhome..??

netmindz, 2023-04-27:
It's likely that you are missing one of the dependencies. Are you using PlatformIO ?

Please send an example of the error

netmindz, 2023-04-27:
Before you spend time on this project, have you confirmed which model of control unit you have? There are a few different projects for the different Balboa systems

mbuhansen, 2023-04-30:
I have now install platformio, and try to program my ESP32, but i cant. when i load your projekt to platformio, press build in bottom i get around 20 errors, it looks like some files is missing to ES32 board. your projekt only want to make firmware to D1mini, it gives error when i connect to ESP32.  I have also try to make new projekt and move your files, but can't get it to work.

netmindz, 2023-05-03:
To help I need to see a screenshot of the errors, the most likely issue is that it's telling you it can't find the wifi.h file with your details

netmindz, 2023-05-03:
https://github.com/netmindz/balboa_GL_ML_spa_control/wiki/Install-Guide

tmjo, 2023-05-03:
Another probable cause is missing libraries SoftwareSerial.h, WebServer.h and similar. I remember it took me some time to find all of them, and the right ones. One drawback when using Arduino IDE is that it stores all dependencies for all projects in one location and makes it hard when one work with multiple projects, and even worse if those projects need different version of a dependency.

I know that PlatformIO does this differently, so that each project has its own dependency. In my opinion it is a much better way. Not sure what is common practice, but I imagined that the specific dependencies used in a PlatformIO project could be added to github so that the project in theory should compile immediately after downloading it.

Do you know what is good practice here @netmindz?

netmindz, 2023-05-03:
Yeah, the "vanilla" Arduino doesn't really have any way of handling project dependencies. Some projects copy 3rd party libraries into their own project, but that's not great. I've tried to use git submodules, but then if people aren't familiar with git they get empty folders.

That's why I swapped the project to PlatformIO, you can then just open and run. OK needs you to install PlatformIO, but if you are a "power" user, of ArduinoIDE, you can just open the sensor/src/sensor.ino file, say yes to rename and fix the deps

mbuhansen, 2023-05-04:
thank you for your answers. here is attached the log from terminal window.
I have open you project, made wifi.h file i src folder. and try to upload to ESP32, at the end i got code wrong chip, it is like ESP32 not is config in your project.

[Platformio error.txt](https://github.com/netmindz/balboa_GL_ML_spa_control/files/11395593/Platformio.error.txt)
[image]

mbuhansen, 2023-05-04:
[Platformio error.txt](https://github.com/netmindz/balboa_GL_ML_spa_control/files/11395633/Platformio.error.txt)

netmindz, 2023-05-04:
You are running the wrong build. The D1 mini is an esp8266 profile. Click on the PlatformIO icon on the left, expand the options for the wemos and select upload and monitor

netmindz, 2023-05-04:
Actually, looks like you are trying to build both targets. 

You definitely only want to be trying to build and upload the target that matches your device.

Why you are seeing compile errors for the dependent library for the esp32 build I'm not sure why.

The project builds correctly in GitHub actions, so there isn't anything magic about my build environment

mbuhansen, 2023-05-04:
Okay, yes i use build button in the bottom of the screen. 
Now i found the upload and monitor in theplatformio menu, but i get stil lots of errors.
[image]
[image]

netmindz, 2023-05-04:
Found the issue @mbuhansen - as you are on windows, it wasn't including the right files

closed 2023-05-04 by netmindz

mbuhansen, 2023-05-04:
Yes i use windows.
But not i works now, app is loaded to ESP32, and can see webserver is online now.
Just to be sure i have to connect pin 19 and 23 from ESP32 to RS485 board.

Many thanks for the help, looking forward to my heating system now can see when the spa heats up...

netmindz, 2023-05-04:
Does the status page now show the mode and temp? If not try swapping a+b on the rs485 or the rx and tx

mbuhansen, 2023-05-04:
Yes status display show up. I have not connect to spa yet, it is the next step.
In mqtt i get error because missing username and password from esp32, how can i write this in code.?

netmindz, 2023-05-04:
Add a #define BROKER_USERNAME and #define BROKER_PASSWORD after BROKER_ADDR

mbuhansen, 2023-05-18:
This it works great. The esp32 is connected and working, i can see all in home assistant. 
Now i just need a little more test, have problem with weak wifi signal at the hottub.
when you send command to hottub , how are you doing this..??

netmindz, 2023-05-18:
Great to hear you have the basic read-only setup working. I have a branch called send-command-queue that allows for toggle of the light and going in between standard and eco mode from home assistant. You will also need to connect up pin5 from the tub to the controller as well as make sure your RS485 adapter is connected up correctly to send as well. Personally, I found that I can only send if it's connected to the 5v not 3.3v supply

mbuhansen, 2023-05-18:
Okay.
Is it the CMD topic i can send message to, if i want to switch mode..???
Do you have a description, how to send command back..??

netmindz, 2023-05-18:
For the features that can be controlled, they appear as a switch within Home Assistant, so no need to know any codes.

Probably best to continue in the discussion section rather than that in this closed bug report 

https://github.com/netmindz/balboa_GL_ML_spa_control/discussions