netmindz/balboa_GL_ML_spa_control #31: Set target temp buggy

netmindz, 2023-06-23:
Sometimes you can use the `climate` HA entry to set the target correctly, sometimes it will change, but misses or duplicates temp+ or temp- actions. You do however see the real target temp in HA so you can try to change again if your first attempt doesn't work as planned

netmindz, 2023-07-23:
How is your experience of setting the temp with your version of the code that only sends the command once @tmjo ?

tmjo, 2023-07-24:
I think it was working alright, just that I have to "manually" update my target setpoint by giving it a up or down (with appropriate delays) to ensure it has the correct target temp. I'll doublecheck to see.

tmjo, 2023-07-31:
I did a lot of testing today, and found out it was not working very well when trying larger steps in temperature. I added a lot of empty command paddings and then it was working fine. Same goes for most of the other commands. Better late than never, I uploaded my adjustments to [a new branch (main-fork-tmjo](https://github.com/netmindz/balboa_GL_ML_spa_control/commit/9522f83f3d930fd6611d7a7945e55cbc7821ba7e) just for info where only the last commit is interesting (the others are temporary stuff I have to do to run it).

Perhaps we could add a generic empty-padding thing instead of adding them all around?

netmindz, 2023-11-05:
I've tweaked the code so that there is a pause between sending commands to better match how a human would be pressing buttons and it's seeming much improved .

Are you able to test how main or the latest release tag works for you @tmjo ?

tmjo, 2023-11-05:
Sure thing, I'll test it out. Can't promise exactly when, it's busy days (and cold+dark outside 😎) but I'll get to it for sure!

netmindz, 2023-11-06:
Can do webOTA in the newer releases so can do some of the work from the comfort of indoors and test with your phone when inside the tub

tmjo, 2023-11-07:
Yeah, I'm mostly doing OTAs but it has failed me one too many times sending me out to physically reboot the ESP. But testing comfortably inside the tub is pretty nice though :)

netmindz, 2023-11-23:
Set temp now working correctly with the pause between commands

closed 2023-11-23 by netmindz