netmindz/balboa_GL_ML_spa_control #50: No web/mqtt on M5 Atom Lite ESP32

Father-Chon, 2023-09-04:
So I've got an M5 Atom Lite ESP32 kicking around (only ESP32's I have) and I've tried to get this working with the June RW build.
 I'm using the same pin numbers https://docs.m5stack.com/en/core/atom_lite

I can telnet (no response from any command or output), port 80 is open but no web page.

I did have to change the platformio type https://docs.platformio.org/en/stable/boards/espressif32/m5stack-atom.html#id1
In the code, the LED part was throwing errors so I had tried both changing to pin 27 as the doc says and then also removing the code for the LED low/highs and still nothing.

```
[env:m5stack-atom]
platform = espressif32
board = m5stack-atom
```
When building, I do get this error, not sure if it's relevant.
```
Compiling .pio\build\m5stack-atom\src\sensor.ino.cpp.o
In file included from .pio/libdeps/m5stack-atom/home-assistant-integration/src/device-types/HAFan.h:5,
                 from .pio/libdeps/m5stack-atom/home-assistant-integration/src/ArduinoHA.h:12,
                 from C:/Users/Shawn/Desktop/balboa_GL_ML_spa_control-0.2.1/sensor/src/sensor.ino:14:
.pio/libdeps/m5stack-atom/home-assistant-integration/src/device-types/../utils/HANumeric.h:232:18: warning: class 'HANumeric' is implicitly friends with itself
     friend class HANumeric;
                  ^~~~~~~~~

```
When I run it from vscode and platformio, upload/monitor I get this

```
Leaving...
Hard resetting via RTS pin...
--- Terminal on COM3 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H

Connecting to IoT.
Connected with IP: 10.0.3.159
Setting pin 22 LOW
Setting serial port as pins 19, 23
Set serial port as pins 19, 23
Configuring WDT...
End of setup
```

Any advice?

netmindz, 2023-09-05:
I'm unsure why the webserver would not be responding for you with that particular chip

If you use the ArduinoIDE and run the webserver example for ESP32 does that work ok

With regards to MQTT, have you set the IP address for your MQTT broker and username & password if required?

How are you checking for messages? Mosquito client?

Father-Chon, 2023-09-05:
ESPhome runs no problem on it (was running on it before). I can try the webserver in a bit. 

Checked MQTT via MQTT explorer and tcpdump listening for any traffic from the IP of the board, nothing.

netmindz, 2023-09-06:
When using tcpdump I take it this is with the broker address set to be the machine you are running tcpdump on?

Father-Chon, 2023-09-06:
Yup, thats correct. I see connections from all other devices on 1883, but nothing from 10.0.3.159 which I confirmed in my router that it is that device and has good wifi signal. 

I have ordered a few Wemos D1 Mini ESP32 anyways, as I wouldn't mind having a few more kicking around, so we'll see if that works out of the box with it. May be something specific to the M5 atom, and since I've already got one set up I'll try and do a bit more debugging, but ESP's are an area I'm pretty new at.

Father-Chon, 2023-09-09:
Confirmed that this works with a Wemos, exact same config just reverting the platformio file back to default, so definitely an issue with the M5 Atom, so other users with other ESP32's may have an issue.

danimaro, 2023-09-10:
I am also experiencing the same problem with my esp32 not connecting to mqtt broker. And I guess this issue can be the reason:
https://github.com/netmindz/balboa_GL_ML_spa_control/issues/51

danimaro, 2023-09-10:
This change solved my problem with both mqtt and webserver:
https://github.com/netmindz/balboa_GL_ML_spa_control/pull/52/commits/e95dfc1d9bb7e9a757eefd8cfa023ede75934289

netmindz, 2023-09-13:
I'll have a think about how to handle returning status when panel select has not been connected correctly

netmindz, 2023-09-13:
Welcome @Father-Chon and @danimaro to the project btw. Good to see more people using it.

Please help spread the word to other hot tub owners.

Being able to control from Home Assistant is so handy. Mine syncs up with time off day electricity prices

danimaro, 2023-09-14:
Awesome project! Helps a lot now that I can control my hot tub after electricity prices

netmindz, 2023-09-14:
Please share info of your setup in https://github.com/netmindz/balboa_GL_ML_spa_control/discussions/24

netmindz, 2023-09-14:
Need to test properly, but should be addressed by https://github.com/netmindz/balboa_GL_ML_spa_control/pull/54

Father-Chon, 2023-09-14:
#54 Seems to have fixed the web interface on the Atom, which is awesome!
I swapped it back out from the Wemos because I wouldn't mind shipping this back, but I've ran in to a couple of issues.

I get no data read, but pin 5 is connecting (not throwing the error) and just says unknown.
If I press an action in home assistant, it keeps sending the same command and cycling it (eg light will turn on and off every 2ish seconds). So it's talking and sending to the hot tub, but there is no data actually coming in.

On the Atom, I've got RX on pin 19, TX on pin 23, nothing to RTS (I am using the recommended ARCELI TTL To RS485 Adapter, my understanding is I don't need this?), and PIN5 to pin 22. Should this be the correct pinout for the Atom? 
These are the docs for the Atom https://docs.m5stack.com/en/core/atom_matrix

netmindz, 2023-09-15:
Sounds like for some reason it can't read data but it can send. One quick thing to try is swapping the A and B connections as data is received, but gibberish.

What does the serial console show? There are a few loggers commented out that can help diagnose what's going on. If things are working correctly you should get multiple FA messages per second.

You can try swapping to a different pins too just in case something is clashing on that particular mcu

netmindz, 2023-09-15:
No need to connect RTS pin if you aren't using the max485

netmindz, 2023-09-15:
Try flipping 19 to 21

netmindz, 2023-09-25:
Any luck getting data read back @Father-Chon ?

Father-Chon, 2023-10-02:
Swapped out the TTL to 485 with the other one I got, tried swapping a and b with no luck. Tried pins TX21 RX25 P5-19 on the atom lite but nothing still. I can still trigger the hot tub. Board is an EL2001 M3 with an ML700 panel on top. Tried swapping the panel to the other port I was using and it works fine, just to verify the port is good. Tried it in the other port and no luck. 

So I can trigger anything from home assistant, but I don't get anything on RX that it can decode. I know someone else that has a similar hot tub and gonna see this week if they've got the same EL2000/2001 and see if it works on theirs.

netmindz, 2023-10-05:
Hi @Father-Chon , that's very useful to know you have the EL rather than GL. As far as I was aware till now they should be cross compatible with this official topside panels, but there might be something different about their output.

In order to help debug, I need to know what information is shown on the serial monitor in Platform IO

If you don't really see anything, then you might need to enable a few more of the Serial.print and Serial.printf lines

If you get lots of messages that say "unknown message length for ff" or similar, then you need to swap your A and B connections.

If after you uncomment the logger that says how many bytes have been read you don't see this at all, then you aren't getting any serial data at all from the tub.

If you are getting series bytes read, not a long stream about message lengths, but not the FA long logger, then your pin5 connection

netmindz, 2023-10-05:
Do you have access to an oscilloscope or multi meter? Does your new RS485 adapter have any status LEDs?

Can you share and photos of your setup?

netmindz, 2023-11-23:
Have you had any further luck @Father-Chon ? What is your current status?

shawnmolnar, 2024-01-15:
No further luck with the M5 Atom Lite, but I ended up keeping the D1. Seems to be something related to the M5 Atom Lite that doesn't like connecting.