netmindz/balboa_GL_ML_spa_control #53: MAX485 Help

davewatson91, 2023-09-13:
Thanks for all the hard work on this one. I've got some time to try and get this going, but am struggling with how to connect up my MAX485 using the existing code - I'm not sure where to connect the additional pins - I couldn't see them defined separately in the code sorry.

netmindz, 2023-09-14:
#define RTS_PIN 22  // RS485 direction control, RequestToSend TX or RX, required for MAX485 board.

That pin needs to be attached to the RE and DE pins

Then the RX and TX connect to the RS485 data pins

davewatson91, 2023-09-15:
Thanks @netmindz 

I've managed to get it going! 

I believe there may be an error in the install instructions currently (wrt wiring). Currently the diagram has the spa Pin 5 going to Pin 22 on the ESP, but that's the MAX485 RE/DE pin, and Pin 18 is setup for the Spa Pin 5. Hopefully I'm interpretting correctly sorry - I'm just trying to get my head around the project. 

So far I haven't managed to get control of the spa working, so that could possibly be a cause. 

Separately, have you got any way of OTA modifying of the ESP code (without a remote host to control it)? Possibly another area where ESPHome can help if not.

davewatson91, 2023-09-16:
Further to this sorry @netmindz, I have identified that it is getting stuck in a loop when trying to command. This could be 100% related to my setup now that I've dug further. 

It's getting stuck in the sendCommand() function, when waiting for a response I believe. My serial monitor just seems to loop and show the "message sent : ..." printf. 

I'm also not recieving a number of different messages (target temp, post temp, cmd, fb etc.) - happy to dig however, but just need some guidance as to where to start. 

[image]

netmindz, 2023-09-16:
Ah I didn't do the diagram. Well spotted.

Are you running the rs485 adapter from a 3.3v supply or 5v? I've had issues where if you use 3.3v then your can receive but not send

netmindz, 2023-09-16:
Target temp can currently only be discovered by pressing the temp button(s) as I grab the value off the screen when you are in that mode.

Some of the others might not populate as I've dialed back on some of the debug to reduce the overhead of sending lots of MQTT but might not have fully removed the definition of that sensor. Some were also "sniffing" what the other panel was sending, but the newer code doesn't listen to the output of the other panel

davewatson91, 2023-09-16:
Thanks @netmindz! 

Currently powering the RS485 adapter (MAX485) through a voltage regulator (set to 5v), feeding from the spa molex. The data pins are set to 3v3 (for the ESP32). The ESP is powered from the same supply. 

Currently it's setup as: 

- spa pin 5 > ESP pin 22
- spa pin 6 > voltage reg positive
- spa pin 8 > voltage reg negative
- spa pin 1 > MAX485 B
- spa pin 3 > MAX485 A
- ESP pin 18 > MAX485 DE
- ESP pin 23 > MAX485 RE
- ESP pin 22 > MAX485 DI && MAX485 RO

Completely understand the other bits with debugging - I might try and go through and identify those bits as I'm going through. Target temp makes sense too, although I don't think it caught mine when it was set by the main panel. 

Once I'm in a known working state, I'll have more of a chance to play and work out what else is going on. Did you have any way of remote dev for this? Or are you physically connecting to the ESP each time? (or using a proxy for it). 

Thanks again!

netmindz, 2023-09-16:
In theory there is ArduinoOTA in the build, but at some point that's stopped working. Not had time to investigate why. While I'm investigating timing related issues like command sending I can't use the telnet logging without affecting timings.

If I restrict to classic esp32 then could try and use two different CPU cores to help offset that, but as I've actually just tried flipping to the LILYGO T-RSC3 setup which gives single board solution, that's only single core. Command sending isn't working for me on that platform

netmindz, 2023-09-16:
Wait a sec, I think you have your max485 connected incorrectly

https://hobbycomponents.com/wired-wireless/663-max485-rs485-transceiver-module

It's the two Enable pins that you contact to RTS

davewatson91, 2023-09-17:
Totally fair with the OTA updates. 

Sorry - I gave my pins from memory - i should have checked beforehand. 

It's setup correctly (I think): 

MAX485 > ESP32
- RO > Pin 19
- DI > Pin 23
- DE & RE > Pin 22

I'm a little bit lost as to what else to try to get it to respond. I don't have another RS485 adapter here to try unfortunately.

netmindz, 2023-09-17:
Which adapter are you currently using @tmjo ?

netmindz, 2023-09-19:
And did you ever have command sending working with that setup? @tmjo

netmindz, 2023-09-19:
This commit *might* help you https://github.com/netmindz/balboa_GL_ML_spa_control/commit/14bbd7a591cbade982c3f7c901f64148ef804cb9 @davewatson91

davewatson91, 2023-09-19:
> This commit _might_ help you [14bbd7a](https://github.com/netmindz/balboa_GL_ML_spa_control/commit/14bbd7a591cbade982c3f7c901f64148ef804cb9) @davewatson91

Thanks @netmindz. I'm away until next week currently, but a quick look at that commit just now - I wasn't getting into that section of code I don't believe. I wasn't ever seeing that serial.printf statement to indicate such.

tmjo, 2023-09-27:
> And did you ever have command sending working with that setup? @tmjo

Can't really remember, but I believe I didn't get that before before I changed to my current setup. It should work though.

netmindz, 2023-10-17:
Hi @davewatson91  - Can you see if the sending of commands now works for you now this fix has been applied. https://github.com/netmindz/balboa_GL_ML_spa_control/commit/e2edeeacd5be723fd308765fd5e26a9eb10ff8d9

davewatson91, 2023-10-17:
> Hi @davewatson91 - Can you see if the sending of commands now works for you now this fix has been applied. [e2edeea](https://github.com/netmindz/balboa_GL_ML_spa_control/commit/e2edeeacd5be723fd308765fd5e26a9eb10ff8d9)

I've just spent a bit of time messing around - for whatever reason, I can't get the TTL RS485 adapter (with automatic control) to work - it just flickers the TX/RX LED's dimly. 

I reverted to the MAX485, and can read the values again, however when I try to make a command, it just loops at the "message sent :  40" line for forever, and when it breaks from it, it hasn't taken effect. 

I've also noticed that there are dip-switch settings for 'P-2', 'P-3' & panel scrunching - has anyone had to change these? 

So overall, unfortunately, still not working 😞

netmindz, 2023-10-18:
Boo. Still stuck as read only for you.

What microcontroller are you using? Classic ESP32 or Pico, C3 or S3?

netmindz, 2023-10-18:
And you pulled the latest code from the main branch?

netmindz, 2023-10-19:
I'm dropping support for the 8266, but hopefully getting closer on an out of the box solution using the LILYGO T-RSC3

davewatson91, 2023-10-20:
> I'm dropping support for the 8266, but hopefully getting closer on an out of the box solution using the LILYGO T-RSC3

I won't bother going down that route then - I'm a little lost as to what to try next. I might need to see if I can source another ESP32.. I've ordered a LILYGO T-RSC3, but no local suppliers, so a long wait on the slow boat unfortunately.

netmindz, 2024-04-28:
Did you manage to buy an ESP32 @davewatson91 ?

davewatson91, 2024-04-28:
> Did you manage to buy an ESP32 @davewatson91 ?

I did - and got it (mostly) working, but shortly thereafter emptied the spa, and haven't had it going again since :( 

I ran into issues with write control, where it would get out of sync, or just lock up if it got too many commands in a row (like bumping the temp up/down).