netmindz/balboa_GL_ML_spa_control #68: Pin5 always shows not detected

Weom85, 2023-11-23:
Hi, first thanks for you great work,

I’m experiencing some issue with detection of pin5, I replace the code to fit the schematic provided so I connected spa pin5 to pin 22, since I’m using the rs485 you recommended I don’t need the RTS so I assign RTS to be pin26.

I get absolutely no data coming from the Spa the RX light on the Rs-485 is blinking all the time and when i send commands the tx is blinking and the command queue seems to send every message as expected.

i measured 4.6v on the pin22 which is connected to spa Pin5 but still I get the status of pin5 not detected. I tried to swap RX TX between RS485 board and ESP32 as well as swapping A and B, still same issue. I follow the schematic you provided. 

Do you have any clue ? 

Thank you

netmindz, 2023-11-23:
Hi @Weom85 

Welcome to the project, how did you find us out of interest?

If you get that status message then we are never seeing the pin you have assigned for pin5 of the spa to be seen as high, so something isn't right of you say that you are seeing around 4.6v between that and ground but the micro controller isn't detecting.

I would double check that you are setting the pin values in the right ifdef block and double checking your connections. We are being a bit cheeky by connecting a 5v input to an MCU that's designed to give only 3.3v output, but it seems to be tolerant even if there is a question mark over official support for that

If you can share the serial monitor output and also try to touch from the 5v or 3.3v pin of the ESP32 to the pin you are using as pin5 to see if you can get the pin5 undetected to disappear at all. If that fails then you definitely are using the wrong pin or might have blown that input

Weom85, 2023-11-24:
Update 

i didn’t had the chance yet to plug it back to the Spa but I troubleshoot the problem on the breadboard, I think my problem is coming the fact that there’s no pull down resistor for the pin5 input. 

Did you guys wired it directly without issue ?

netmindz, 2023-11-24:
I have no pulldown, but as you are seeing the pin as always LOW, then I don't see how INPUT_PULLDOWN or external pulldown would help

netmindz, 2023-11-24:
What microcontroller are you using? Don't forget to share the serial monitor debug info

Weom85, 2023-11-24:
I’m using a ESP32, I was troubleshooting on my bench yesterday and I changed the code to print the variable panelSelect and panelDetect and I saw that even if I was not connected to the Spa I received a true value, I then add a status message pin5 detected in the If statement where you set the panelSelect to true. So if my pin5 the signal is unstable maybe that’s why I’m not getting new information and maybe that’s why the only status I got is pin5 not detected.

I will try again to connect on the spa and I’ll change the wiring to be able to connect the computer to take the log of serial console without frying the MCU.

Weom85, 2023-11-24:
I just realized that with that status update in the void loop I’ll not get any updated status. Since I know now that the pin5 is assign to the good IO I’ll revert to the released version and give it a try

Weom85, 2023-11-25:
Hi, i just install fresh version of the code and check with the serial log and this what i got : 

Configuring WDT...
End of setup
Button press - Down
                   ERROR: Pin5 went high before command could be sent after flus                                                                                        h
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command before flush : 20
                                                      ERROR: Pin5 went high befo                                                                                        re command could be sent after flush
Switch Light changed - Toggle
ERROR: Pin5 went high before command before flush : 20
                                                      ERROR: Pin5 went high befo                                                                                        re command could be sent after flush
Switch Light changed - Toggle
ERROR: Pin5 went high before command before flush : 20
                                                      ERROR: Pin5 went high befo                                                                                        re command could be sent after flush
onPumpSwitchStateChanged Aux 0
                              Sending fb0603450e000af5c0 - 1 times
                                                                  ERROR: Pin5 we                                                                                        nt high before command before flush : 20
                                        ERROR: Pin5 went high before command cou                                                                                        ld be sent after flush
Button press - Time
                   ERROR: Pin5 went high before command before flush : 20
                                                                         ERROR:                                                                                         Pin5 went high before command could be sent after flush

and the web interface shows that :

Mode: -1
State: Panel select (pin5) not detected
Temp: -1.0C
Heating: false
Last Update: Sat Nov 25 2023 12:01:37 GMT-0500 (heure normale de l’Est nord-américain)

Weom85, 2023-11-25:
I used my logic analyzer and i get these reading, if it can help to debug

[image]
[image]

netmindz, 2023-11-25:
Looks like we are missing some of the initial messages as we only see the end of the setup, not the earlier messages that show which pins are assigned.

Have you been able to try my suggested test to confirm the pin you are trying to use to detect pin5 is actually working?

Weom85, 2023-11-25:
I didn’t modify the program that time I use the original and I connected PIN 5 to GPIO 18, but on the graph we can see that the PIN5 goes low when a message arrive but there’s a lot of spike on PIN5 and I don’t know if it can cause the problem

Weom85, 2023-11-28:
I don’t know if you read this message directly on GitHub or in your email but I added graph snapshot that I took with my logic analyser. You might have to check the message directly on GitHub web page to see them. My board look like a balboa 54302, top panel looks like a 107071. I was able to get a FA message and see the temperature in the bytes so I’m pretty sure the problem it’s not the spa, like I said pin 5 drop for 5ns every 8.3 ms maybe that have something to do with the problem. I was connected to my own wifi.

netmindz, 2023-11-28:
Looking at the graph, The longer period of LOW that are when the FA is sent and time for you to send the FB back looks normal but definitely not the regular periods where it goes low for only a really brief pulse

Looking at the part number you list for the control board. Looks like you might have the EL2000 series, which I **_assume_** should be cross-compatible - given the topside panels look to support either EL or GL 2000/2001 and we are basically emulating the ML500 panel

netmindz, 2023-11-28:
Also coming up as COLEMAN/MAAX by Balboa, possibly buyout?

netmindz, 2023-11-28:
I don't know if a small capacitor might help suppress those false readings?

Weom85, 2023-11-28:
I’ll try to add a capacitor but it need to be very small I believe , I’ll let you know as soon as I can test that I’m not home for the next few days.

At least you confirm that the rest of the behaviour is normal so as Soon I fix the 8ns drop I should be good.

Weom85, 2023-12-06:
@netmindz this is the one I have Analyseur logique professionnel durable 5 MHz 24 m 8 canaux mini avec câbles pour l'industrie https://a.co/d/iS9TyjU

closed 2023-12-07 by Weom85

netmindz, 2023-12-07:
How did you fix the issue with the low spikes on pin5? Capacitor? Some other hardware tweak? @Weom85