netmindz/balboa_GL_ML_spa_control #69: Support for Fahrenheit

Weom85, 2023-12-07:
It seems that the actual code is not working when tub is set in Fahrenheit

Weom85, 2023-12-07:
I changed few line to get the temp to show properly but I think it would be nice if the code could check which unit is sent by the spa and adjust accordingly 

[image]

Now I get information but I’m unable to send command :(

netmindz, 2023-12-07:
There is some code that handles reading the value no matter if C or F, but the tricky part is that at the moment the unit of measurement is sent in the setup method, which is before we have read any data from the hottub. Feel free to send a PR that adds a build flag to enable this to be set at build time.

Weom85, 2023-12-08:
The problem is the code will not execute properly since one of the first IF statement is looking for “43” or “2d” that’s why the code was not working for me, and for me it was 46

netmindz, 2023-12-08:
Sorry, my mistake. I thought we had both 43 and 46, we must have lost that bit of code somewhere along the lines

netmindz, 2023-12-08:
yeah, we have some handling of Fahrenheit in the standalone library, I've just tweaked the main code too so at least you can read data, it will just show the wrong symbol in Home Assistant unless you edit the two references to

setUnitOfMeasurement("°C");

closed 2023-12-08 by netmindz

netmindz, 2023-12-08:
I've added a new USE_FAHRENHEIT flag too so that HA looks correct