netmindz/balboa_GL_ML_spa_control #71: Not able to send command

Weom85, 2023-12-09:
[image]
Hi, im not able to send command , when i check with analyser it seems that the delay between the FA message and the FB message is too long and pin 5 goes HI before i finish to send message.

i added log and snapshot of the graph.

do you have any idea of what i should change or what i'm doing wrong

100.000000
           Sent target temp data 100.000000
                                           onPumpSwitchStateChanged Pump1 2
                                                                           Sendi                                                                                        ng fb060343060006f9a2 - 1 times
                               ERROR: Pin5 went high before command could be sen                                                                                        t after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
ERROR: Pin5 went high before command could be sent after flush
YAY: command response : 3350
                            Switch Light changed - Toggle
ERROR: Pin5 went high before command could be sent after flush

Weom85, 2023-12-09:
[image]
When the spa is sending the FB message there's 300 micro second between FA and FB

netmindz, 2024-01-03:
I've optimised the code a bit to try and decrease the delay between when the interrupt on pin5 triggers and we send the command, if you can try pulling the latest code.

Overall I'm not sure why you were seeing a longer delay than I do, but hopefully this will help a bit more.

Generally if I complete sending and see the response in around 2500 then my spa responds ok

Weom85, 2024-01-03:
Thank you ! I’ll give it a try I was able to make it work tweaking the code but I lost connection often right now it’s not really fully reliable hopefully with your new code will fix it :)

I also order a RSC3 but i cannot flash the chip platformio won't compile the code.

I will still try to work on that

netmindz, 2024-01-06:
Can you remember what optimisations you made to make it more reliable?

As for the RSC3 I don't know if my board is faulty, but not only was I seeing it be slower, but even when it said it had send a message there was no response.

I need to pull that board out and confirm that it is actually transmitting to the rs485 bus when it says it is