netmindz/balboa_GL_ML_spa_control #72: Pin5 Problem

comanche3, 2023-12-20:
My new ESP32 isn´t connecting. It always shows Pin5 problem, allthough it is showing allready correct temperature. My Pin5 is connected to ESP32-Nr.22 as shown in manual.
In the sensor.cpp it says
#define RTS_PIN_DEF 22  // RS485 direction control, RequestToSend TX or RX, required for MAX485 board.
#define PIN_5_PIN_DEF 18

So i would say it must be connected to Pin 18 instead ?? What is correct ?

netmindz, 2023-12-26:
The RTS pin is only need of you use the older style rs485 adapter like the max485

If you use the recommended adapter then not required.

Pin 5 of the GL controller must be connected to the pin you define in the config, which looks like pin 18 you are using

If you have further issue, please share the full output of the webpage and what text is shown on the serial monitor

netmindz, 2024-02-11:
Did you resolve your problems @comanche3 ?