netmindz/balboa_GL_ML_spa_control #74: Fahrenheit showing up as Celcius

shawnmolnar, 2024-01-15:
Now that I've got my issues sorted out with the M5 Atom Lite, now I'm running in to another issue where the hot tub is reporting in Fahrenheit, but it's showing up as celcius.

I'm using the latest main branch, and turned on the USE_FAHRENHEIT flag, but in home assistant I'm still reporting as 10.1C when it should be 101F. Drop that flag, change hot tub to Celcius, and it works fine, but wife's preference is Fahrenheit. 
Looks like there needs to be some additional handling in the tub temp side of things for F vs C. 

I'm no developer, but I may take a stab at it if I have some time coming up. For now I'm in Celsius until I can get this side figured out. 

`    hvac.setMinTemp(26);
    hvac.setMaxTemp(40);
    hvac.setTempStep(0.5);`
`                if (result.substring(10, 12) == "43" || result.substring(10, 12) == "46") {  // "C" or "F"
                    double tmp = (HexString2ASCIIString(result.substring(4, 10)).toDouble() / 10);
                    if (menu == "46") {
                        tubTargetTemp = tmp;
                        targetTemp.setValue((float)tubTargetTemp);
                        if (sendBuffer.isEmpty()) {  // supress setting the target while we are changing the target
                            hvac.setTargetTemperature((float)tubTargetTemp);
                        }
                        Serial.printf("Sent target temp data %f\n", tubTargetTemp);
                    } else {
                        if (tubTemp != tmp) {
                            tubTemp = tmp;
                            temp.setValue((float)tubTemp);
                            hvac.setCurrentCurrentTemperature((float)tubTemp);
                            Serial.printf("Sent temp data %f\n", tubTemp);
                        }
                        if (heaterState && (tubTemp < tubTargetTemp)) {
                            double tempDiff = (tubTargetTemp - tubTemp);
                            float timeToTempValue = (tempDiff * MINUTES_PER_DEGC);
                            timeToTemp.setValue(timeToTempValue);
                        } else {
                            timeToTemp.setValue((float) 0);
                        }
                    }`

netmindz, 2024-01-15:
I've never tried swapping the unit of measurement after it was created in Home Assistant. 

You might need to delete the device and add it again or maybe just the two entities

Weom85, 2024-01-21:
on that line :  double tmp = (HexString2ASCIIString(result.substring(4, 10)).toDouble() / 10);

just remove the divided by 10 it will fix the 10.1 and in Home assistant you can change the unit for F