netmindz/balboa_GL_ML_spa_control #77: Code not compiling

chrisguen, 2024-07-02:
Hey, I got some errors while compiling, I didnt change anything except creating the wifi_credentials.h and changing the board. 
[image]

netmindz, 2024-07-02:
Please share your platformio.ini file, looks like there is an issue with your setup.

What microcontroller are you trying to use?

chrisguen, 2024-07-03:
`; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]

default_envs = nodemcu-32s

[env]
monitor_speed = 115200
framework = arduino
lib_deps = 
	https://github.com/dawidchyrzynski/arduino-home-assistant.git#2.0.0
	links2004/WebSockets@^2.3.7
	https://github.com/EinarArnason/ArduinoQueue.git@^1.2.5
	https://github.com/netmindz/ESP-WebOTA.git#disable-interrupts

[env:wemos_d1_mini32]
platform = espressif32
board = wemos_d1_mini32
monitor_filters = esp32_exception_decoder

[env:nodemcu-32s]
platform = espressif32
board = nodemcu-32s
monitor_filters = esp32_exception_decoder
[...]`

netmindz, 2024-07-03:
That's just a regular ESP32 so no need to change the config

Are you sure you are building with platformio and no a generic c++ build?

chrisguen, 2024-07-03:
Stupid mistake from me, I just cloned the repo and didnt use the release.

closed 2024-07-03 by chrisguen