netmindz/balboa_GL_ML_spa_control #78: Infinite command retry

netmindz, 2024-07-09:
A new thread to cover the issue that the command retry causes infinite retry for some users

netmindz, 2024-07-09:
So  - the genuine top panels actually respond to every FA message with an FB, but most of the time this is just an IDLE message - more details here https://github.com/netmindz/balboa_GL_ML_spa_control/discussions/3

Some users have reported that as long as we delay sending FB messages with commands, they can send commands and them be accepted.
Others have tried doing as the genuine panel does and send IDLE commands when there is no command to send.

Personally I have not managed to get 100% acceptance of command sending if transmitted only once - which when you are trying to build up complex sequences of commands for Home Assistant automation rather than just toggling the pump from your mobile phone can cause lots of issues. 

Therefore the builds I use and the current code in the repo will send a command and keep trying to send if either the data was not transmitted before pin5 changed state and will log an error to the serial log or until it sees a change in the state of the hot tub as reflected in the FA message. As I never use IDLE messages, therefore any command should result in a state change, e.g toggle light command should mean a different state of the light. 

Where this fails is you try and send a command that is unexpected by the controller and therefore ignored, e.g sending the time command for a controller that has it's DIP switches set to not expect time based scheduling. The controller will just keep ignoring the unexpected command.

I have developed the code so far largely based on my own top panel, which is an unbranded 3rd party controller designed to be compatible with the ML5xx range, with the GL2000 configured for 2 single-speed pumps dedicated circulation pump.

For users of other configurations, we might need to send different commands, e.g I'm assuming that given there is only ever a single button for a pump, the panel always sends the same command but the controller cycles between all possible states depending on single/dual config.

If the toggle of the light works as expected, but others do not then we need to capture the FB command sent by your genuine panel and see how the FA value changes in response to this command

netmindz, 2024-07-11:
This is the version of the code that should give logging of all the messages, including the FB commands sent by your own panel if I remember correctly 

https://github.com/netmindz/balboa_GL_ML_spa_control/tree/v0.0.9-readonly

netmindz, 2024-07-11:
You can just read the raw stream of serial data too using RS485 to USB adapter and view data as hex, then look for the FA and FB sequences within that, but it's a bit easier to use the sketch to see each messages one by one

JamieEC, 2024-07-16:
Hi,
As per our email convo I am experiencing this issue. I have a ML700 panel with a GL2000 controller.

I have tried to analyse the responses by printing the result variable from the handleMessage function but I cannot see any changes when a button is pressed (other than the initial mode change to temp set). Could you explain how the dequeue works (lastraw3 variable etc.)? Is the updated set temp supposed to be sent once an up/down button is pressed?

I have had limited success by decommenting lines 952 & 954, however this seems to be largely luck based as commands are still missed.

I will try that readonly script at some point to see what I can analyse further, or provide a capture to be analysed by those with more knowledge than me.

This is also affecting setting a pump speed for me as sometimes it will press the jets button twice. I am not 100% sure if it is related as I have a 2 speed pump, which I am not sure if you have tested.

Also having issues with the mode dropdown for standard, eco, etc. where the function is not called when moving from eco to std. I can start a new thread for this if it is unrelated as I suspect.

Thanks for all your work on this so far.